triton-droids/embedded #2: Set up env for hardware

jakenorbie12, 2025-02-01:
Set up the environment so that when the hardware parts comes we can run programs on them and test them.

danielsanei, 2025-02-02:
**UPDATE (Jan - Week 4)**
Set up Ubuntu environment (_Ubuntu Desktop 24.04.1 LTS (64-bit)_) for general use.

danielsanei, 2025-02-23:
UPDATE (Feb - Week 3)
Successfully connected to Pi5 remotely via SSH.

Command used: ssh triton-droids@192.168.113.206

Considerations: Need to be on the same network, However, UCSD-Protected requires using a team member's personal UCSD login credentials. Also, UCSD-GUEST supports a wide range of devices and potentially has security restrictions on devices accessing each other, and therefore is not a viable backup (tested this, resulted in connection timed out). Tested this temporarily with a local network, which worked successfully.

[image]

danielsanei, 2025-03-03:
UPDATE (Feb - Week 3)
Successfully installed ROS2 (_Jazzy Jalisco_). This version is a Long Term Support (LTS) release, meaning it has extended support including updates and bug fixes through May 2029.

Main Installation Guide Used: https://automaticaddison.com/how-to-install-ros-2-jazzy/
Supporting Video: https://www.youtube.com/watch?v=ZGds6NuZLzo

Fun Fact: This is the 10th release of ROS2, and was released in May 2024.

[image]

kevshin2002, 2025-04-04:
Individual hardware needs to be integrated manually and we won't know unless we know the sensor / device.

Some devices already have open source ros integration codebases available online, but it's hard to make nodes without knowing the devices we are using.

danielsanei, 2025-04-26:
UPDATE: TODO

closed 2025-10-21 by danielsanei