triton-droids/embedded #4: Fix faulty SD

jakenorbie12, 2025-02-01:
Fix a faulty flash system in our hardware.

danielsanei, 2025-02-02:
**UPDATE (Jan - Week 4)**
SD card originally flashed successfully and operated as expected. After re-boot, Ubuntu desktop would not load, giving the following error:

[image]

Attempted the following fixes on two different computers:

1. Erase and re-format SD card (Rasberry Pi Imager, Balena Etcher, Diskpart CMD)
2. Re-flash SD card (Rasberry Pi Imager, Balena Etcher) using MicroSD adapter, USB adapter, raw PC port

All failed, indicating corrupted SD card.

Provided my personal 64 GB SanDisk flashed with Ubuntu as a temporary substitute, until we are able to acquire a replacement from Amazon.

kevshin2002, 2025-04-11:
Do we still have the corrupted SD card?

i can try to take a look at it.

danielsanei, 2025-04-12:
Hey Kevin, Uday was able to order a replacement SD card so we have a new one that is no longer corrupted. Currently, we are using my personal SD card. I tried to copy that image and move it to the new, Triton Droids SD card, but I've run into issues with the copy process due to the image size. Both SD cards are 64 GB, but it seems the current image is slightly larger than what the new SD card has room for. I tried using a few different softwares, and trying to flash a ZIP copy rather than the raw, but unfortunately I ran into the same issue every time. Do you have any ideas for new solutions I could try? I would love to give this another shot and start using the club's new SD card!

danielsanei, 2025-04-19:
UPDATE (APRIL - Week 3)
Successfully flashed new SD card, restored files, updated system with ROS2 Jazzy (all steps are just as done before, no changes). My personal SD card can now be replaced with the Triton Droids SD card.

[image]

closed 2025-04-19 by danielsanei