triton-droids/embedded #6: Run code on Pi

jakenorbie12, 2025-04-07:
Short Description of Task: Run any code on the raspberry pi, either wired or wireless, as a proof of concept.

How it Helps: Shows progress and is required for all other tasks.

End Goal: Run any code on the raspberry pi, either wired or wireless, as a proof of concept.

GustavoGarcia-1, 2025-04-12:
Tried doing SSH but the IP keeps changing of the RPI because of UCSD-PROTECTED. Trying to solve the issue right now.

GustavoGarcia-1, 2025-04-12:
If it doesnt work, we will try using the ethernet cable and make a wired connection.

GustavoGarcia-1, 2025-04-20:
Tailscale isn't setup up on the pi with the new sd card because of limited access to the triton droids github account. Once it is Daniel and I will work on creating the simple circuit. Then we will give the pi to Tyler to work on wired ssh.

GustavoGarcia-1, 2025-04-26:
Daniel and I met on Wednesday to run code and get a circuit working.
-We discovered that to keep everything organized, it’s best to use pip (not sudo) when downloading libraries. pip installs libraries locally, while sudo installs them system wide. Since multiple people will be using the Pi, we want to keep things clean and avoid system wide changes. This came up while installing VS Code and the necessary libraries.
-We worked in a virtual environment, as it ensures all commands and installations remain local.
-We wrote a simple LED on/off script in Python, but faced library related errors.

smhitle, 2025-05-12:
Now able to flash an LED on the Pi using gpiozero which is apparently the recommended library and has python specific documentation. Used GPIO17 (pin 11) specifically.

The command to run: sudo python3 pythonled.py

The following is the physical setup for flashing an LED on the Pi.

- male to female wire from GPIO17 to positive terminal
- male to female wire from GND (pin 6) to negative terminal
- male to male from positive terminal to positive pin of LED
- transistor from negative pin of LED to negative terminal NOTE: brown side should be connected to negative terminal

[image]

closed 2025-05-17 by jakenorbie12