triton-droids/embedded #7: Run python code on Pi

jakenorbie12, 2025-04-07:
Short Description of Task: Run some python code, as simple as you want, on the raspberry pi and have it sense sensors and affect motors.

How it Helps: All work that is to be done involving sensors and actuators needs this as the baseline.

End Goal: Run some python code, as simple as you want, on the raspberry pi and have it sense sensors and affect motors.

kevshin2002, 2025-04-09:
#6 ?

jakenorbie12, 2025-04-09:
> [#6](https://github.com/triton-droids/embedded/issues/6) ?

To clarify, this is running the most simple program imaginable, such as turning on a light or making a buzzer make noise. For running python the next objective would be to make motors or sensors move using python in a high level manner, with abstraction and objects.

parthtr9, 2025-04-12:
We made the ssh work by using tailscale. Here is the documentation on how to ssh into a mac:
Installing Tailscale on Mac

1. Go to this website: https://tailscale.com/download 
2. Choose download for mac (.pkg ver)
3. Open the pkg and complete the setup
4. Once downloaded: 
5. Go to system settings → general → login items and extensions → Allow tailscale to run in the background & in the ‘Network extension’ under “Extension” allow tailscale.
6. Quit tailscale and run it again
7. In tailscale settings allow CLI integration 
8. Log into tailscale using your github acc, join the Triton-droids org team
9. You should be on the network now
10. Through your terminal run `tailscale up` then run `tailscale status`
11. You should see the ips of every device on tailscale; look for `tritondroids@`
12. ssh using the ip
13. Ex. ssh triton-droids@100.27.82.10
14. Enter the password (td1024) when prompted

smhitle, 2025-04-13:
SSH Into RPI using Tailscale for WSL on Windows

1. Follow these steps to download Tailscale on WSL
https://tailscale.com/kb/1039/install-ubuntu-2004

2. Once you run "sudo tailscale up", authenticate using your GitHub account

3. Select "triton-droids.org.github" and Connect

4. Once you are running tailscale, run sudo tailscale status to list out the devices on the network

5. Connect to the RPI by running ssh tritondroids@<ip of RPI>

6. Use the password td1024 to login

closed 2025-05-17 by jakenorbie12