triton-droids/embedded #9: Run Crane Assisted Flexibilty Test on Robot

jakenorbie12, 2025-04-07:
Short Description of Task: Either through own code or from simulation, run a flexibilty test on the lower body robot with the assistance of a crane. This will need the lower body to be assembled first.

How it Helps: It is important both as a demo/proof of concept, and also is important to ensure the specification set by mechanical are correct for range of motion.

End Goal: Either through own code or from simulation, run a flexibilty test on the lower body robot with the assistance of a crane.

kevshin2002, 2025-04-09:
Not an embedded task, but simulation.

jakenorbie12, 2025-04-09:
It is running code on the robot once made

kevshin2002, 2025-04-09:
How far are we looking into for these sprints?

We don't have any mechanical or hardware to work off of.

jakenorbie12, 2025-04-09:
An optimistic view of the quarter. Most of these issues are sequential.